jaretburkett/ILI9488 #3: TFT LCD ILI9488 is not displaying correct image

titusece, 2017-12-06:
Hello,
I am trying to interface the ILI9488 (320x480) LCD display into ARM board which boots Linux OS.
And using FBTFT driver for that.
https://github.com/notro/fbtft
I have created a driver for ILI9488 based on the above driver base and also I have added your init code.
Now I can see some thing on my LCD display but not correct image and you mentioned that ILI9488 doesn't supports 16bit mode (but data sheet mentioned that it supports) but 24bit mode.

What could be the problem ?
Can you please help me out on this ?

https://www.dropbox.com/s/d4lpuefnl98j473/problem-warp-display.webm?dl=0

Thanks!

jaretburkett, 2017-12-14:
This has nothing to do with this library.

closed 2017-12-14 by jaretburkett

titusece, 2017-12-16:
Please refer to the below link for fix.
https://github.com/notro/fbtft/issues/490#issuecomment-350557428