jaretburkett/ILI9488 #14: Pinmasks and ports not defined for esp32

maxymoo22, 2020-07-04:
When I compile your example sketch for an arduino uno, it works fine. However when I switch the board type to esp32 dev module, I get a bunch of errors:

```
C:\Users\Max\Documents\Arduino\sketches\libraries\ILI9488-master\ILI9488.cpp: In member function 'void ILI9488::spiwrite(uint8_t)':

C:\Users\Max\Documents\Arduino\sketches\libraries\ILI9488-master\ILI9488.cpp:115:3: error: 'mosiport' was not declared in this scope

  *mosiport |=  mosipinmask;

   ^

C:\Users\Max\Documents\Arduino\sketches\libraries\ILI9488-master\ILI9488.cpp:115:16: error: 'mosipinmask' was not declared in this scope

  *mosiport |=  mosipinmask;

                ^

C:\Users\Max\Documents\Arduino\sketches\libraries\ILI9488-master\ILI9488.cpp:118:3: error: 'mosiport' was not declared in this scope

  *mosiport &= ~mosipinmask;

   ^

C:\Users\Max\Documents\Arduino\sketches\libraries\ILI9488-master\ILI9488.cpp:118:16: error: 'mosipinmask' was not declared in this scope

  *mosiport &= ~mosipinmask;

                ^

C:\Users\Max\Documents\Arduino\sketches\libraries\ILI9488-master\ILI9488.cpp:121:8: error: 'clkport' was not declared in this scope

       *clkport |=  clkpinmask;

        ^

C:\Users\Max\Documents\Arduino\sketches\libraries\ILI9488-master\ILI9488.cpp:121:20: error: 'clkpinmask' was not declared in this scope

       *clkport |=  clkpinmask;

                    ^

exit status 1
Error compiling for board ESP32 Dev Module.
```

Is `clkpinmask` some sort of built in thing for a regular arduino? Because I don't see where it's defined so yeah. Any help on this would be greatly appreciated. Thanks.

Pconti31, 2021-07-19:
@maxymoo22 It's defined inside his ILI9488.cpp routine.

```
#if defined(USE_FAST_PINIO) && !defined (_VARIANT_ARDUINO_STM32_)
    clkport     = portOutputRegister(digitalPinToPort(_sclk));
    clkpinmask  = digitalPinToBitMask(_sclk);
```
Appears the defines are not setup for esp32.

kc9zda, 2021-08-07:
So I encountered the same problem today, and here's how I solved it:

In the ILI9488::spiwrite function, there's a line that looks like this...
`#if defined(ESP8266) || defined (ARDUINO_ARCH_ARC32)`

that needs to be changed to this...
`#if defined(ESP8266) || defined (ARDUINO_ARCH_ARC32) || defined(ESP32)`

so the compiler chooses the block of code that uses the digitalWrite function instead of the block of code with the pinmasks. The example sketch worked for me once I changed this one line.