jaretburkett/ILI9488 #16: ILI9488 not working on Arduino Mega

mattbue, 2020-12-24:
Hey,
i have connected an ILI9488 to an Arduino Mega using the pins of the HW SPI.
There is nothing shown on the display unless i press the reset button of the Arduino Mega.
As long the reset button is pressed the last "picture" is displayed. When the button is released the screen is cleared and keeps cleared until the button is pressed again.

Does someone has an idea what can cause this behaviour?

Thanks

ihsanbu, 2022-01-09:
/***************************************************
  This is our GFX example for the Adafruit ILI9488 Breakout and Shield
  ----> http://www.adafruit.com/products/1651
3.5 Inch 480x320 SPI Serial TFT LCD Module Display Screen with Press Panel Driver IC ILI9488 for MCU

  Check out the links above for our tutorials and wiring diagrams
  These displays use SPI to communicate, 4 or 5 pins are required to
  interface (RST is optional)
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/

#include "SPI.h"
#include <Adafruit_GFX.h>
#include <ILI9488.h>

//#define TFT_CS         PA1
//#define TFT_DC         PB3
//#define TFT_LED        PB0
//#define TFT_RST        PB4

#define TFT_CS         10
#define TFT_DC         9
//#define TFT_LED      PB0
#define TFT_RST        8

// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
ILI9488 tft = ILI9488(TFT_CS, TFT_DC, TFT_RST);
// If using the breakout, change pins as desired
//Adafruit_ILI9488 tft = Adafruit_ILI9488(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);

void setup() {
  Serial.begin(9600);
  Serial.println("ILI9488 Test!");

// delect all SPI Devics
   pinMode(TFT_CS,OUTPUT);
   pinMode(7,OUTPUT);
   digitalWrite(TFT_CS,HIGH);
   digitalWrite(7,HIGH);
 
  tft.begin();

  // read diagnostics (optional but can help debug problems)
  uint8_t x = tft.readcommand8(ILI9488_RDMODE);
  Serial.print("Display Power Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9488_RDMADCTL);
  Serial.print("MADCTL Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9488_RDPIXFMT);
  Serial.print("Pixel Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9488_RDIMGFMT);
  Serial.print("Image Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9488_RDSELFDIAG);
  Serial.print("Self Diagnostic: 0x"); Serial.println(x, HEX);

  Serial.println(F("Benchmark                Time (microseconds)"));

  Serial.print(F("Screen fill              "));
  Serial.println(testFillScreen());
  delay(500);

  Serial.print(F("Text                     "));
  Serial.println(testText());
  delay(3000);

  Serial.print(F("Lines                    "));
[... 265 lines cut ...]
  w     = min(tft.width(), tft.height());
  start = micros();
  for(i=0; i<w; i+=6) {
    i2 = i / 2;
    tft.drawRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(i, 0, 0));
  }

  return micros() - start;
}

unsigned long testFilledRoundRects() {
  unsigned long start;
  int           i, i2,
                cx = tft.width()  / 2 - 1,
                cy = tft.height() / 2 - 1;

  tft.fillScreen(ILI9488_BLACK);
  start = micros();
  for(i=min(tft.width(), tft.height()); i>20; i-=6) {
    i2 = i / 2;
    tft.fillRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(0, i, 0));
  }

  return micros() - start;
}

ihsanbu, 2022-01-09:
please save libraries with name that header file name
my code is working best

mr-baguette, 2024-05-15:
> /*************************************************** This is our GFX example for the Adafruit ILI9488 Breakout and Shield ----> http://www.adafruit.com/products/1651 3.5 Inch 480x320 SPI Serial TFT [...]

@ihsanbu 
i tried this code but it didn't work for me, but im not sure if all the correct pins are connected. If someone could explain to me how i would connect this code to my screen, that would be great. All my available pins are : sdo(miso), led, sck, sdi(mosi), dc/rs, reset and cs. 
and unlike MattBue s problem, there is just nothing displayed, even when holding the reset button