jaretburkett/ILI9488 #19: not working with stm32DUINO

rohitlucky, 2021-04-22:
hi
can anyone help,i am getting error in compling the example graphictest.

ILI9488.h:186:14: error: 'RwReg' does not name a type
  186 |     volatile RwReg *mosiport, *clkport, *dcport, *rsport, *csport;
      |              ^~~~~
exit status 1
Error compiling for board Generic STM32F1 series.

djcleckie, 2022-01-15:
Hi

I have the same error on a STM 64 Nucleo F401RE board.
Look about line 35 of the header file/
typedef volatile uint32 RwReg;
change it to typedef  uint32_t RwReg;

unit32 and unit32_t seems they are not quite the same.

Also this affects it
#ifdef ARDUINO_STM32_FEATHER
typedef volatile uint32 RwReg;
#endif
Now if you are using a different STM32 board this line is ignored.

These changes helped but I am still not managing to compile.

TheHuziek, 2023-05-06:
try to change the ili9488.h file and move the typedef volatile uint32_t RwReg; out of ifdef