jaretburkett/ILI9488 #23: Bad performance - Unoptimized 16-bit color to 24 bit(18 bit) conversion.

nickshl, 2023-04-13:
This is not a good way to do such conversion:
`  uint8_t r = (color & 0xF800) >> 11;
  uint8_t g = (color & 0x07E0) >> 5;
  uint8_t b = color & 0x001F;

  r = (r * 255) / 31;
  g = (g * 255) / 63;
  b = (b * 255) / 31;`
Much better(2.3 times actually) to do it as such:
`    // Create 24-bit RGB values from 16 bit
    uint8_t r = (color & 0xF800) >> 8; // >> 11 << 3
    uint8_t g = (color & 0x07E0) >> 3; // >> 5 << 2
    uint8_t b = (color & 0x001F) << 3;`
But still painfully slow for full screen output

username7291, 2023-09-29:
I think he does not maintain it anymore. I found out that too. The thing is for every type of drawing write16BitColor is called. And this color Conversion is done, fillscreen gets 7 times faster if you calculate the color once in Fillrect and only use in the loop writergb color.
 void ILI9488::writeRGBColor(uint8_t r, uint8_t g, uint8_t b) {
  spiwrite(r);
  spiwrite(g);
  spiwrite(b);
}

Usind rgb would better for the user anyway always search the color you want in a internet calculator. But this require to change adafruit_gfx too cause any screen text comes from there.

username7291, 2023-09-29:
--global--
uint8_t rgb[3];

void ILI9488::writeRGBColor(uint8_t r, uint8_t g, uint8_t b) {
rgb[0]=r;
rgb[1]=g;
rgb[2]=b;
SPI.transfer(&rgb, 3);
}

Using one time Color conversion in Rect/Lines and this new Version gives 2-10 times faster (works for Atmega328p only no support for other boards or Software SPI).
Thats original Version on a Arduino Pro Mini 8MHZ:
Screen fill              82842504
Text                     1036360
Lines                    14257784
Horiz/Vert Lines         6661184
Rectangles (outline)     3652224
Rectangles (filled)      204570448
Circles (filled)         15878368
Circles (outline)        6078896
Triangles (outline)      3176008
Triangles (filled)       62845464
Rounded rects (outline)  4412440
Rounded rects (filled)   199001768

Thats the optimized Version:

Screen fill              9371376
Text                     498496
Lines                    9162976
Horiz/Vert Lines         759088
Rectangles (outline)     434904
Rectangles (filled)      22852544
Circles (filled)         2920160
Circles (outline)        3871624
Triangles (outline)      1768400
Triangles (filled)       8477368
Rounded rects (outline)  1372496
Rounded rects (filled)   22776720

You could optimize this more, for fillrect/screenfill for example write more pixels at once if the free Memory allows it.