jaretburkett/ILI9488 #24: i use ili9488 to arduino nano anda i connect rts 13,cs 12,rs 11. after i try the example from the ili9488 library, the lcd just white screen. i need a help how to fix this,please thank you

brahmazstra, 2023-04-15:


brahmazstra, 2023-04-15:
this is my setup pin.  i connect my rs pin to dc pin, are they have same function?

#include "SPI.h"
#include <Adafruit_GFX.h>
#include <ILI9488.h>

#define TFT_CS         12
#define TFT_DC         11//my lcd pin RS
#define TFT_RST        13
ILI9488 tft = ILI9488(TFT_CS, TFT_DC, TFT_RST);

username7291, 2023-10-03:
Use the wireing like in the example. Arduino SPI Pins are defined in Hardware you cant use 10/11/12/13 like you want.

#define TFT_CS         10
#define TFT_DC         9
#define TFT_RST        12

SCK is 13, 11 is Data.

marcinoess, 2025-04-15:
Logika ILI9488 działa na 3.3V tak samo zasilanie oraz LED, zrób dzielniki napięcia z rezystorów dodatkowo przylutuj rezystor ceramiczny 104 do zasilania przy samej płycie głównej wyświetlacza. Arduino--------2k-------pin_ILI9488---------3.3k-------GND
Miałem ten sam problem, to pomogło.pozdrawiam z POLSKI